stereolabs/zed-ros-examples #3: Zed examples repo must be added to workspace after zed-ros-wrapper

thatrobotguy, 2020-02-14:
So I have started looking at the ROS versions of the ZED code base and trying to compile the code. I have found that this code repo does not compile normally when you add the `zed-ros-warpper` repo to the same ROS workspace. This repo must be added to the ROS workspace **after** the `zed-ros-wrapper` repo has been compiled in the workspace. If you don't do this, you start to see this error:
```
[  8%] Built target _zed_wrapper_generate_messages_check_deps_objects
Scanning dependencies of target zed_obj_det_sub
[  9%] Building CXX object zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/src/zed_obj_det_sub_tutorial.cpp.o
Generating reconfiguration files for Zed in zed_wrapper
Wrote header file in /home/devmachine2/ros1_ws/devel/include/zed_wrapper/ZedConfig.h
[  9%] Built target zed_wrapper_gencfg
Scanning dependencies of target dynamixel_sdk
[  9%] Building CXX object dynamixel/CMakeFiles/dynamixel_sdk.dir/src/dynamixel_sdk/packet_handler.cpp.o
[ 10%] Building CXX object dynamixel/CMakeFiles/dynamixel_sdk.dir/src/dynamixel_sdk/protocol1_packet_handler.cpp.o
[ 10%] Building CXX object dynamixel/CMakeFiles/dynamixel_sdk.dir/src/dynamixel_sdk/protocol2_packet_handler.cpp.o
[ 11%] Building CXX object dynamixel/CMakeFiles/dynamixel_sdk.dir/src/dynamixel_sdk/group_sync_read.cpp.o
/home/devmachine2/ros1_ws/src/zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/src/zed_obj_det_sub_tutorial.cpp:27:10: fatal error: zed_wrapper/object_stamped.h: No such file or directory
 #include <zed_wrapper/object_stamped.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/build.make:62: recipe for target 'zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/src/zed_obj_det_sub_tutorial.cpp.o' failed
make[2]: *** [zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/src/zed_obj_det_sub_tutorial.cpp.o] Error 1
CMakeFiles/Makefile2:6410: recipe for target 'zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/all' failed
make[1]: *** [zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
I was wondering why this is an issue since the code file throwing the error (and its corresponding CMake file) has the `zed-wrapper` package dependency. I did not directly see the `object_stamped.h` file, but there is a `object_stamped.msg` file available. Why is this an issue?

Myzhar, 2020-02-14:
Hi @thatrobotguy 
this repository requires that the `zed-ros-wrapper` has been installed and compiled before you can compile. 
When you compile the `zed_wrapper` package the `catkin_make` command generates the `object_stamped.h` file from the `object_stamped.msg` descriptor.

thatrobotguy, 2020-02-14:
That makes sense
I just don't understand why the `zed-ros-examples` package is trying to compile _at the same time_. It should be compiled _after_ the other package. Isn't there a way to do that in the package.xml file or `CMakelists.txt` file to wait to compile? I am pretty sure there is...

Myzhar, 2020-02-14:
I need to check if all the dependencies are correctly written for all the examples. Stay tuned for oncoming updates.

thatrobotguy, 2020-02-14:
sounds like a plan

Myzhar, 2020-02-18:
Solved with https://github.com/stereolabs/zed-ros-examples/commit/5474c90b6bdde6e711ae7f6bc6d0523be7f3b860

thatrobotguy, 2020-02-18:
I just tried to compile with the simple fix, and I still get that same issue. 
Here is the error after running `catkin_make` for the second time:
```
[ 43%] Generating EusLisp manifest code for sick_scan
[ 44%] Generating Python code from SRV zed_wrapper/reset_tracking
Scanning dependencies of target zed_wrapper_generate_messages_lisp
/home/devmachine2/ros1_ws/src/zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/src/zed_obj_det_sub_tutorial.cpp:27:10: fatal error: zed_wrapper/object_stamped.h: No such file or directory
 #include <zed_wrapper/object_stamped.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[ 44%] Built target sick_scan_generate_messages_py
zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/build.make:62: recipe for target 'zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/src/zed_obj_det_sub_tutorial.cpp.o' failed
make[2]: *** [zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/src/zed_obj_det_sub_tutorial.cpp.o] Error 1
[ 44%] Generating Python code from SRV zed_wrapper/start_object_detection
CMakeFiles/Makefile2:6410: recipe for target 'zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/all' failed
make[1]: *** [zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/CMakeFiles/zed_obj_det_sub.dir/all] Error 2
[ 45%] Generating Lisp code from zed_wrapper/object_stamped.msg
make[1]: *** Waiting for unfinished jobs....
[ 45%] Built target sick_scan_generate_messages_cpp
[ 46%] Generating Python code from SRV zed_wrapper/start_3d_mapping
[ 46%] Generating Lisp code from zed_wrapper/objects.msg
```

Myzhar, 2020-02-18:
The `zed_wrapper` should be indeed compiled as first since all the `package.xml` of this repository contain the field `<depend>zed_wrapper</depend>`, but for some reason this is not enough.
I added a new dependency also in CMakeFiles.txt with commit https://github.com/stereolabs/zed-ros-examples/commit/0b4341a6481607cb3a6bc4472d3633106b802645.

Can you pull and test the new commit since it seems that your configuration replicates the issue in any case?

thatrobotguy, 2020-02-18:
That appears to have done the trick. I have not used that `CMake` syntax before so I did not think to add that. Thanks for the help man.

closed 2020-02-18 by thatrobotguy