stereolabs/zed-ros-examples #8: Using ROS integration with multiple ZED?

poorboy44, 2020-03-26:
Is it possible to use the ROS integration with multiple ZED cameras? The documentation seems to indicate so, but when I tried to do so, it didn't allow me (launching the second camera caused the first camera to close/abort).

Myzhar, 2020-03-26:
@poorboy44 you cannot run a node with the same name in the same namespace, to start the second node you must change the node name (if they must stay in the same namespace) or you must change the namespace.

If you are using the ZED launch files the namespace is the same as the camera name:
For example:
`roslaunch zed_wrapper zed.launch camera_name:=zedB`

Instead to change the node name:
`roslaunch zed_wrapper zed.launch node_name:=zed_node_B`

poorboy44, 2020-03-27:
I tried that, and it didn't seem to work. The second camera is unable to launch for some reason. 

In the first terminal, I launch the first camera without a problem.

```
(base) poorboy44@poorboy44-XPS-8930:~/catkin_ws$ roslaunch zed_wrapper zed2.launch serial_number:=27696585 node_name:=new_zed camera_name:=new_zed
... logging to /home/poorboy44/.ros/log/4096e6bc-6fd9-11ea-8f25-5cea1d90be79/roslaunch-poorboy44-XPS-8930-2918.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://poorboy44-XPS-8930:40965/

SUMMARY
========

PARAMETERS
 * /new_zed/new_zed/auto_exposure_gain: True
 * /new_zed/new_zed/auto_whitebalance: True
 * /new_zed/new_zed/brightness: 4
 * /new_zed/new_zed/contrast: 4
 * /new_zed/new_zed/depth/depth_downsample_factor: 1.0
 * /new_zed/new_zed/depth/depth_stabilization: 1
 * /new_zed/new_zed/depth/max_depth: 20.0
 * /new_zed/new_zed/depth/min_depth: 0.7
 * /new_zed/new_zed/depth/openni_depth_mode: 0
 * /new_zed/new_zed/depth/quality: 3
 * /new_zed/new_zed/depth/sensing_mode: 0
 * /new_zed/new_zed/depth_confidence: 50
 * /new_zed/new_zed/depth_texture_conf: 100
 * /new_zed/new_zed/exposure: 100
 * /new_zed/new_zed/gain: 100
 * /new_zed/new_zed/gamma: 8
 * /new_zed/new_zed/general/base_frame: base_link
 * /new_zed/new_zed/general/camera_flip: False
 * /new_zed/new_zed/general/camera_model: zed2
 * /new_zed/new_zed/general/camera_name: new_zed
 * /new_zed/new_zed/general/gpu_id: -1
 * /new_zed/new_zed/general/grab_frame_rate: 30
 * /new_zed/new_zed/general/resolution: 2
 * /new_zed/new_zed/general/self_calib: True
 * /new_zed/new_zed/general/svo_compression: 2
 * /new_zed/new_zed/general/verbose: False
 * /new_zed/new_zed/general/zed_id: 0
 * /new_zed/new_zed/hue: 0
 * /new_zed/new_zed/mapping/fused_pointcloud_freq: 1.0
 * /new_zed/new_zed/mapping/mapping_enabled: False
 * /new_zed/new_zed/mapping/max_mapping_range: -1
 * /new_zed/new_zed/mapping/resolution: 0.05
 * /new_zed/new_zed/object_detection/confidence_threshold: 50
 * /new_zed/new_zed/object_detection/object_tracking_enabled: True
 * /new_zed/new_zed/object_detection/od_enabled: False
 * /new_zed/new_zed/object_detection/people_detection: True
 * /new_zed/new_zed/object_detection/vehicle_detection: False
 * /new_zed/new_zed/point_cloud_freq: 15.0
 * /new_zed/new_zed/pos_tracking/area_memory: True
 * /new_zed/new_zed/pos_tracking/area_memory_db_path: 
 * /new_zed/new_zed/pos_tracking/fixed_z_value: 0.0
 * /new_zed/new_zed/pos_tracking/floor_alignment: False
 * /new_zed/new_zed/pos_tracking/imu_fusion: True
 * /new_zed/new_zed/pos_tracking/init_odom_with_first_valid_pose: True
 * /new_zed/new_zed/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /new_zed/new_zed/pos_tracking/map_frame: map
 * /new_zed/new_zed/pos_tracking/odometry_frame: odom
 * /new_zed/new_zed/pos_tracking/path_max_count: -1
 * /new_zed/new_zed/pos_tracking/path_pub_rate: 2.0
 * /new_zed/new_zed/pos_tracking/publish_map_tf: True
 * /new_zed/new_zed/pos_tracking/publish_tf: True
 * /new_zed/new_zed/pos_tracking/two_d_mode: False
 * /new_zed/new_zed/pub_frame_rate: 15.0
[... 304 lines cut ...]
[ INFO] [1585279299.694197628]:  * [DYN] saturation		-> 4
[ INFO] [1585279299.694368637]:  * [DYN] sharpness		-> 4
[ INFO] [1585279299.694539420]:  * [DYN] gamma			-> 8
[ INFO] [1585279299.694686512]:  * [DYN] auto_exposure_gain	-> ENABLED
[ INFO] [1585279299.695091873]:  * [DYN] auto_whitebalance	-> ENABLED
[ INFO] [1585279299.697333045]:  * Camera coordinate system	-> Right HANDED Z UP and X FORWARD
[ INFO] [1585279299.697361382]:  *** Opening ZED...
[ INFO] [1585279300.964569775]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279304.303267556]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279307.599037171]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279310.904646302]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279314.222635764]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279317.516227772]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279320.800369212]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279324.103824622]: ZED connection -> CAMERA STREAM FAILED TO START
[ZED]ERROR: sl::Camera::Open has not been called, no Camera instance running.
[ INFO] [1585279327.471401387]: ZED connection -> CAMERA STREAM FAILED TO START
```

Myzhar, 2020-03-27:
Sorry @poorboy44, you are right. You must not only change `camera_name` and/or `node_name`, but you must discriminate the two cameras using their `camera_id` or `serial_number`.

There is an example that helps you:
https://github.com/stereolabs/zed-ros-examples/tree/master/examples/zed_multicamera_example

You can start two cameras and Rviz with the command:
`$ roslaunch zed_multicamera_example zed_multicamera.launch`
that uses a modified launch file that allows you to correctly change all the required parameters:
https://github.com/stereolabs/zed-ros-examples/blob/master/examples/zed_multicamera_example/launch/zed_camera_mod.launch

closed 2020-06-01 by Myzhar