stereolabs/zed-ros-examples #10: Cannot run zed_rtabmap.launch

saeedarabi92, 2020-03-28:
Hello,
I have a zed2 camera and could not run rtabmap example on my Jetson TX2. The rtabmap freezes and does not show anything.

**logs and files** 

```roslaunch zed_rtabmap_example zed_rtabmap.launch -v
... logging to /home/nvidia/.ros/log/d34fb552-7131-11ea-a696-00044bc73598/roslaunch-nvidia-desktop-32725.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://nvidia-desktop:44019/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /zed/rtabmap/Grid/3D: True
 * /zed/rtabmap/Grid/CellSize: 0.05
 * /zed/rtabmap/Grid/ClusterRadius: 0.1
 * /zed/rtabmap/Grid/DepthDecimation: 1
 * /zed/rtabmap/Grid/DepthRoiRatios: [0.0, 0.0, 0.0, 0.0]
 * /zed/rtabmap/Grid/FlatObstacleDetected: True
 * /zed/rtabmap/Grid/FootprintHeight: 2.0
 * /zed/rtabmap/Grid/FootprintLength: 0.18
 * /zed/rtabmap/Grid/FootprintWidth: 0.18
 * /zed/rtabmap/Grid/FromDepth: True
 * /zed/rtabmap/Grid/GroundIsObstacle: False
 * /zed/rtabmap/Grid/MapFrameProjection: False
 * /zed/rtabmap/Grid/MaxGroundAngle: 30.0
 * /zed/rtabmap/Grid/MaxGroundHeight: -0.4
 * /zed/rtabmap/Grid/MaxObstacleHeight: 0.1
 * /zed/rtabmap/Grid/MinClusterSize: 3
 * /zed/rtabmap/Grid/MinGroundHeight: -0.5
 * /zed/rtabmap/Grid/NoiseFilteringMinNeighbors: 5
 * /zed/rtabmap/Grid/NoiseFilteringRadius: 0.1
 * /zed/rtabmap/Grid/NormalK: 20
 * /zed/rtabmap/Grid/NormalsSegmentation: False
 * /zed/rtabmap/Grid/PreVoxelFiltering: True
 * /zed/rtabmap/Grid/RangeMax: 3.0
 * /zed/rtabmap/Grid/RangeMin: 0.7
 * /zed/rtabmap/Grid/RayTracing: True
 * /zed/rtabmap/GridGlobal/Eroded: False
 * /zed/rtabmap/GridGlobal/FootprintRadius: 0.18
 * /zed/rtabmap/GridGlobal/FullUpdate: True
 * /zed/rtabmap/GridGlobal/MaxNodes: 0
 * /zed/rtabmap/GridGlobal/MinSize: 1.0
 * /zed/rtabmap/GridGlobal/OccupancyThr: 0.55
 * /zed/rtabmap/GridGlobal/ProbClampingMax: 0.971
 * /zed/rtabmap/GridGlobal/ProbClampingMin: 0.1192
 * /zed/rtabmap/GridGlobal/ProbHit: 0.7
 * /zed/rtabmap/GridGlobal/ProbMiss: 0.4
 * /zed/rtabmap/GridGlobal/UpdateError: 0.01
 * /zed/rtabmap/Mem/IncrementalMemory: true
 * /zed/rtabmap/Mem/InitWMWithAllNodes: false
 * /zed/rtabmap/approx_sync: True
 * /zed/rtabmap/config_path: ~/.ros/rtabmap.cfg
 * /zed/rtabmap/database_path: ~/.ros/rtabmap.db
 * /zed/rtabmap/frame_id: base_link
 * /zed/rtabmap/map_frame_id: map
 * /zed/rtabmap/odom_frame_id: odom
 * /zed/rtabmap/odom_sensor_sync: False
 * /zed/rtabmap/odom_tf_angular_variance: 0.001
 * /zed/rtabmap/odom_tf_linear_variance: 0.001
 * /zed/rtabmap/publish_tf: True
 * /zed/rtabmap/queue_size: 10
 * /zed/rtabmap/scan_normal_k: 0
 * /zed/rtabmap/subscribe_depth: True
[... 545 lines cut ...]
        <remap from="odom"              to="$(arg odom_topic)"/>

        <!-- localization mode -->
        <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
        <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
        <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/>
    </node>

</launch>

```

**rtabmap version**

```nvidia@nvidia-desktop:~/.ros$ rosversion rtabmap
0.19.3
nvidia@nvidia-desktop:~/.ros$ rosversion rtabmap_ros
0.19.3
```

rviz and rtabviz images:
[image]
[image]

Thank you.

Myzhar, 2020-03-30:
Hi @saeedarabi92 
thank you for reporting the problem. It will be fixed as soon as possible.

dxdpmdx, 2020-05-12:
i have the same issue with latest code:
roslaunch zed_rtabmap_example zed_rtabmap.launch
... logging to /home/nvidia/.ros/log/c6c117cc-941d-11ea-8410-00044bcbee62/roslaunch-miivii-tegra-11760.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://127.0.0.1:44643/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /zed/rtabmap/Grid/3D: True
 * /zed/rtabmap/Grid/CellSize: 0.05
 * /zed/rtabmap/Grid/ClusterRadius: 0.1
 * /zed/rtabmap/Grid/DepthDecimation: 1
 * /zed/rtabmap/Grid/DepthRoiRatios: [0.0, 0.0, 0.0, 0.0]
 * /zed/rtabmap/Grid/FlatObstacleDetected: True
 * /zed/rtabmap/Grid/FootprintHeight: 2.0
 * /zed/rtabmap/Grid/FootprintLength: 0.18
 * /zed/rtabmap/Grid/FootprintWidth: 0.18
 * /zed/rtabmap/Grid/FromDepth: True
 * /zed/rtabmap/Grid/GroundIsObstacle: False
 * /zed/rtabmap/Grid/MapFrameProjection: False
 * /zed/rtabmap/Grid/MaxGroundAngle: 30.0
 * /zed/rtabmap/Grid/MaxGroundHeight: -0.4
 * /zed/rtabmap/Grid/MaxObstacleHeight: 0.1
 * /zed/rtabmap/Grid/MinClusterSize: 3
 * /zed/rtabmap/Grid/MinGroundHeight: -0.5
 * /zed/rtabmap/Grid/NoiseFilteringMinNeighbors: 5
 * /zed/rtabmap/Grid/NoiseFilteringRadius: 0.1
 * /zed/rtabmap/Grid/NormalK: 20
 * /zed/rtabmap/Grid/NormalsSegmentation: False
 * /zed/rtabmap/Grid/PreVoxelFiltering: True
 * /zed/rtabmap/Grid/RangeMax: 3.0
 * /zed/rtabmap/Grid/RangeMin: 0.7
 * /zed/rtabmap/Grid/RayTracing: True
 * /zed/rtabmap/GridGlobal/Eroded: False
 * /zed/rtabmap/GridGlobal/FootprintRadius: 0.18
 * /zed/rtabmap/GridGlobal/FullUpdate: True
 * /zed/rtabmap/GridGlobal/MaxNodes: 0
 * /zed/rtabmap/GridGlobal/MinSize: 1.0
 * /zed/rtabmap/GridGlobal/OccupancyThr: 0.55
 * /zed/rtabmap/GridGlobal/ProbClampingMax: 0.971
 * /zed/rtabmap/GridGlobal/ProbClampingMin: 0.1192
 * /zed/rtabmap/GridGlobal/ProbHit: 0.7
 * /zed/rtabmap/GridGlobal/ProbMiss: 0.4
 * /zed/rtabmap/GridGlobal/UpdateError: 0.01
 * /zed/rtabmap/Mem/IncrementalMemory: true
 * /zed/rtabmap/Mem/InitWMWithAllNodes: false
 * /zed/rtabmap/approx_sync: True
 * /zed/rtabmap/config_path: ~/.ros/rtabmap.cfg
 * /zed/rtabmap/database_path: ~/.ros/rtabmap.db
 * /zed/rtabmap/frame_id: base_link
 * /zed/rtabmap/map_frame_id: map
 * /zed/rtabmap/odom_frame_id: odom
 * /zed/rtabmap/odom_sensor_sync: False
 * /zed/rtabmap/odom_tf_angular_variance: 0.001
 * /zed/rtabmap/odom_tf_linear_variance: 0.001
 * /zed/rtabmap/publish_tf: True
 * /zed/rtabmap/queue_size: 10
 * /zed/rtabmap/scan_normal_k: 0
 * /zed/rtabmap/subscribe_depth: True
 * /zed/rtabmap/subscribe_odom_info: False
 * /zed/rtabmap/subscribe_rgbd: False
 * /zed/rtabmap/subscribe_scan: False
 * /zed/rtabmap/subscribe_scan_cloud: False
[... 352 lines cut ...]

[image]

git log:
commit e4b925f41ae08838b0ae8ccfa624080af395418a (HEAD -> master, origin/master, origin/HEAD)
Author: Walter Lucetti <walter.lucetti@stereolabs.com>
Date:   Mon Mar 30 09:46:41 2020 +0200

    Fix RTABmap example (#12)

commit ecfd996ee3b0471771ea7c25f7733d45b8489be3
Author: Walter Lucetti <walter.lucetti@stereolabs.com>
Date:   Sun Mar 29 13:42:45 2020 +0200

    Delete zed_nodelet_laserscan.launch

commit 7e9e62d64e28ff300b7c71b5a9968f6c0692341c
Author: Walter Lucetti <walter.lucetti@stereolabs.com>
Date:   Wed Mar 25 22:30:13 2020 +0100

    Update zed_obj_det_sub_tutorial.cpp

commit 2f6b555a2b571cf0702effbb019670bab90d96d9
Author: Walter Lucetti <walter.lucetti@stereolabs.com>
Date:   Fri Mar 20 20:04:45 2020 +0100

Myzhar, 2020-05-12:
Hi @dxdpmdx can you please copy and paste the result of the `roswtf` command while running the nodes?

dxdpmdx, 2020-05-12:
nvidia@miivii-tegra:~$ roswtf
Loaded plugin tf.tfwtf
No package or stack in the current directory
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take a while...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules
running tf checks, this will take a second...
... tf checks complete

Online checks summary:

Found 1 warning(s).
Warnings are things that may be just fine, but are sometimes at fault

WARNING The following node subscriptions are unconnected:
 * /zed/zed_state_publisher:
   * /zed/joint_states
 * /zed/rtabmapviz:
   * /zed/goal_node
 * /zed/rtabmap:
   * /zed/goal_node
   * /zed/global_pose
   * /zed/imu
   * /zed/gps/fix
   * /zed/initialpose
   * /zed/user_data_async
   * /zed/goal
 * /zed/rviz:
   * /zed/map_updates

Found 1 error(s).

ERROR TF multiple authority contention:
 * node [/zed/zed_node] publishing transform [odom] with parent [map] already published by node [/zed/rtabmap]
 * node [/zed/rtabmap] publishing transform [odom] with parent [map] already published by node [/zed/zed_node]

Myzhar, 2020-05-12:
Ok, the problem is with the `odom` topic.
Have you updated both the `zed-ros-examples` and the `zed-ros-wrapper` repositories to the latest version and recompiled?

dxdpmdx, 2020-05-12:
yes, i have updated both of the repositories to the latest version and recompiled, but the issue still exists

Myzhar, 2020-05-12:
OK, I'm checking it

dxdpmdx, 2020-05-12:
thank you very much

Myzhar, 2020-06-01:
Fixed with commit https://github.com/stereolabs/zed-ros-examples/pull/16/commits/487085c840e2a1792bd8e3bb2753b8784768399c

closed 2020-06-01 by Myzhar

smokemonkey101, 2024-09-20:
i See there is a rtabmap with the ROS Noetic
but i cant find any support for rtabmap on ROS2
**How do i run RTABmap on ROS2 with zed**

[zed-ros-examples]  https://github.com/stereolabs/zed-ros-examples
has examples/zed_rtabmap_example

[zed-ros2-examples]  https://github.com/stereolabs/zed-ros2-examples
only has 2   zed_aruco_localization  and  zed_depth_to_laserscan

Myzhar, 2024-09-23:
Hi @saeedarabi92
please:
1. never reply to a closed ticket, always create a new one and link the relevant existing tickets
2. this is a repository for ROS, not ROS 2
3. this is not a bug, but a question, so please use the [correct place](https://community.stereolabs.com)  to ask your questions

smokemonkey101, 2024-09-23:
all good points thank you

PR #12 (2020-03-30): Fix RTABmap example
Fix #10