stereolabs/zed-ros-examples #9: No such file or directory  #include <zed_interfaces/ObjectStamped.h>

saeedarabi92, 2020-03-28:
Hi,

When I tried to compile the zed-ros-examples with my Jetson TX2, received this error:

`/home/nvidia/Aria/src/zed-ros-examples/tutorials/zed_obj_det_sub_tutorial/src/zed_obj_det_sub_tutorial.cpp:28:10: fatal error: zed_interfaces/ObjectStamped.h: No such file or directory
 #include <zed_interfaces/ObjectStamped.h>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.`

What I am I missing?

Thanks

Myzhar, 2020-03-28:
Hi @saeedarabi92 
You have surely updated the `zed-ros-wrapper` repository, but not the `zed-ros-examples` repository.
Try this:
`roscd zed_examples`
`git pull`
Then try to recompile

saeedarabi92, 2020-03-28:
This solved my problem. Thanks. Closing this.

closed 2020-03-28 by saeedarabi92

kuzhang, 2021-06-29:
Hi Myzhar, I face the same issues. I have tried your suggestion, but it didn't work. Also,  I cannot find the .h files inside the zed_interface packages. What did I miss?  Thank you.

> Hi @saeedarabi92
> You have surely updated the `zed-ros-wrapper` repository, but not the `zed-ros-examples` repository.
> Try this:
> `roscd zed_examples`
> `git pull`
> Then try to recompile

kuzhang, 2021-06-29:
> @kuzhang before compiling try to delet the `build` and `devel folders in your catkin workspace to clean the CMake cache

Thanks for the prompt reply. I had tried to clean the WS. It didn't work in my case. I found out it is the rosdep problem. Before I had run the 
```shell
rosdep install --from-paths src --ignore-src -r -y
```

I feel like the rosdep will look at the wrong place.

So I reinstall the rosdep with 

```shell
rosdep init
```
and skip the rosdep install , and execute the 
```shell
catkin_make -DCMAKE_BUILD_TYPE=Release
``` 
and the compiling success.