stereolabs/zed-ros-examples #13: ZED2 is not capable to track AR tags with zed_ar_track_alvar

LucaRoma97, 2020-05-08:
Hi everyone!
I'm working on my thesis about an autonomous charging system. 
I bought a ZED2 camera and a Jetson Nano. After having installed the SDK of the ZED and ROS (Melodic version), I cloned both the zed-ros-examples ([](https://github.com/stereolabs/zed-ros-examples)) and zed-ros-wrapper  ([]((https://github.com/stereolabs/zed-ros-wrapper)).) I also change the zed_wrapper .launch file inside the zed_ar_track_alvar.launch (zed2.launch in place of zed_camera.launch). But when I try to launch the zed_ar_track_alvar.launch, the _Rviz visualizations_ show me warnings and errors and I can't see the output of the camera.
[image]
If I try to visualize the _rqt_graph_:
[image]
Then, I tried to change the _Image Topic_ of Left Image, Left Camera and Tags  ,as shown below:
[image]
Now I can visualize the output of the camera, but it is not yet capable to track the ar codes and the rqt_graph now is the following:
[image]
It seems that the node zed_ar_track_alvar doesn't communicate with the camera ones.
I tried to change several names to the topics without any results. Someone has some advices for tracking the ar tags?
Thank you in advance.

LucaRoma97, 2020-05-09:
I've just fixed the Camera model error, but it still doesn't work.

ArghyaChatterjee, 2020-05-31:
@LucaRoma97 how have you fixed the camera model error?

Myzhar, 2020-06-01:
Fixed with https://github.com/stereolabs/zed-ros-examples/commit/0e924d870e73ef778e73afa948e3c56f87cb84ab

closed 2020-06-01 by Myzhar