stereolabs/zed-ros-examples #17: "rosrun zed_tracking_sub_tutorial zed_tracking_sub" gives nothing back

cnpcshangbo, 2020-07-02:
Hi,
I have followed the steps to install ROS wrapper and example.
The ZED 2 node works wells. I am able to use rviz to check topics.
However, when I run the tracking sub, it gives nothing back.
Hope to get your suggestions.
Thanks,
Bo

Myzhar, 2020-07-02:
Hi @cnpcshangbo 
have you enabled the Object Detection module setting the parameter `object_detection/od_enabled` in `zed2.yaml` to `true` or calling the `start_object_detection` service?

You can read more in the [online documentation](https://www.stereolabs.com/docs/ros/object-detection/)

Myzhar, 2020-07-02:
Hi @cnpcshangbo 
the tutorial has been written with the "original zed" in mind and it does not take in consideration the fact that the "zed mini" and the "zed2" use different namespaces.

To correctly get the output you must remap the topics to match the correct names:
`$ rosrun zed_tracking_sub_tutorial zed_tracking_sub /zed/zed_node/odom:=/zed2/zed_node/odom /zed/zed_node/pose:=/zed2/zed_node/pose`

Let me know if this works for you.

cnpcshangbo, 2020-07-02:
Hi @Myzhar 
It works for me.
Thanks.
Bo

closed 2020-07-02 by cnpcshangbo