stereolabs/zed-ros-examples #22: Change resolution for depth and color

ruthvik92, 2020-11-05:
I am trying to change the resolution of the camera for color and depth.Is there any way to do this in the launch file? Any tutorials available for similar modifications?

Myzhar, 2020-11-05:
Hi @ruthvik92 
to change the grab resolution of the ZED camera you can change the parameter `resolution` in the file `common.yaml`:
https://github.com/stereolabs/zed-ros-wrapper/blob/master/zed_wrapper/params/common.yaml#L26
If you do not want to change the YAML file you can set it in the launch file (`zed.launch`, `zedm.launch` or `zed2.launch`) but I do not recommend it because many modifications are required taking this way

ManChrys, 2021-01-10:
@Myzhar except common.yaml can we change params on /zed-ros-examples/examples/zed_rtabmap_example/params/rtabmap.yaml?

One more question: 

is there in difference if this yaml files say zed or zed2 ? because i have zed2 and i didnt change the model which is zed.

Myzhar, 2021-01-10:
If you use RTABmap you must change the parameters in that yaml file according to your configuration, also the camera model

ManChrys, 2021-01-10:
@Myzhar 

thank you my friend

closed 2021-01-11 by Myzhar