stereolabs/zed-ros-examples #23: [Question] Zed_rtabmap.launch

ManChrys, 2021-01-10:
Hello guys,

im running the launch file and worked fine but while im exporting the pointcloud or when im doing <<edit-->view point cloud>>  then crashes.

[image]

im following the path but the log file doesnt appear to exist.

[image]

I dont know what im doing wrong.Any suggestion? :P 

Thanks in advance.

Myzhar, 2021-01-10:
Hi @ManChrys 
This is a problem with RTABmap not related to the ZED wrapper. Try asking on the RTABmap repo:
 https://github.com/introlab/rtabmap_ros

ManChrys, 2021-01-10:
Ok thank you my friend.

closed 2021-01-10 by ManChrys