stereolabs/zed-ros-examples #24: [Question] Finer pointcloud

ManChrys, 2021-01-11:
Hello guys,

when im using zed_rtabmap_example i have this result

[image]

Post processing makes it a little better but still have that "noise". Which parameters improve the pointcloud?

Thanks again guys.

Myzhar, 2021-01-11:
Hi @ManChrys 
this is another question to post on the RTABmap repository/forum.
Those are the better places to get more precise replies about how RTABmap works

ManChrys, 2021-01-11:
oh ok. i miss understood it because the project is in this repo. My bad then. Thank you again my friend.

closed 2021-01-11 by ManChrys