stereolabs/zed-ros-examples #26: Catkin_make compilation error

Alataishi, 2021-03-04:
I cloned both zed_ros_wrapper and zed_ros_examples, and when I enter catkin_make it shows a lot of error in the files zed_od_display.cpp, zed_od_display.hpp, zed_od_info.cpp and zed_od_info.hpp

[image]

[image]

Myzhar, 2021-03-04:
Hi @Alataishi 
I need more information to help you to solve your issue.
What is the version of the ZED SDK that you installed? What is your OS? What versions of ROS Wrapper and ROS Examples have you cloned?

Alataishi, 2021-03-04:
@Myzhar Hello 
I am actually new to ROS , but anyway the am using Ubuntu 16.04, 
ZED SDK for Ubuntu 16 version 3.4.1
I cloned the the zed wrapper and examples from master branch

Alataishi, 2021-03-04:
it turns out that i have to enable C++11

closed 2021-03-04 by Alataishi