stereolabs/zed-ros-examples #27: [BUG] RTAB_MAP LAUNCH ERROR

dineshrboson, 2021-03-11:
[image]

continously closig

Myzhar, 2021-03-11:
Hi @dineshrboson 
I just tested the example and it's working as expected.
Can you provide me with more information about your system configuration to be able to understand what can be the cause of your problem?

dineshrboson, 2021-03-11:
System:    Host: APUBUDP03 Kernel: 5.4.0-66-generic x86_64 bits: 64
           Desktop: Gnome 3.28.4 Distro: Ubuntu 18.04.5 LTS
Machine:   Device: desktop System: Gigabyte product: Gaming B8 serial: N/A
           Mobo: Gigabyte model: Gaming B8-CF v: x.x serial: N/A
           UEFI: American Megatrends v: F7 date: 07/06/2017
CPU:       Quad core Intel Core i5-7500 (-MCP-) cache: 6144 KB
           clock speeds: max: 3800 MHz 1: 3603 MHz 2: 3599 MHz 3: 3599 MHz
           4: 3599 MHz
Graphics:  Card: NVIDIA GP107 [GeForce GTX 1050 Ti]
           Display Server: x11 (X.Org 1.20.8 )
           drivers: nvidia (unloaded: modesetting,fbdev,vesa,nouveau)
           Resolution: 1920x1080@60.00hz
           OpenGL: renderer: GeForce GTX 1050 Ti/PCIe/SSE2
           version: 4.6.0 NVIDIA 460.32.03
Audio:     Card-1 Intel 200 Series PCH HD Audio driver: snd_hda_intel
           Card-2 NVIDIA GP107GL High Def. Audio Controller
           driver: snd_hda_intel
           Sound: Advanced Linux Sound Architecture v: k5.4.0-66-generic
Network:   Card: Intel Ethernet Connection (2) I219-V driver: e1000e
           IF: enp0s31f6 state: down mac: e0:d5:5e:0b:74:74
Drives:    HDD Total Size: 1240.3GB (4.4% used)
           ID-1: /dev/sda model: KINGSTON_SA400S3 size: 240.1GB
           ID-2: /dev/sdb model: ST1000DM010 size: 1000.2GB
Partition: ID-1: / size: 219G used: 48G (24%) fs: ext4 dev: /dev/sda4
           ID-2: /opt size: 915G used: 1.8G (1%) fs: ext4 dev: /dev/dm-0
           ID-3: /tmp size: 6.0G used: 112M (2%) fs: ext4 dev: /dev/sdb1
           ID-4: swap-1 size: 1.02GB used: 0.79GB (78%)
           fs: swap dev: /dev/dm-1
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 50.0C mobo: N/A gpu: 62C
           Fan Speeds (in rpm): cpu: N/A
Info:      Processes: 332 Uptime: 18:09 Memory: 4097.4/15968.1MB
           Client: Shell (bash) inxi: 2.3.56

Myzhar, 2021-03-11:
Many information, but not what I expected. My fault, I had to be more precise.
What ROS version are you using? Are you using the latest versions of bor the zed-ros-wrapper and zed-ros-examples repositories?
I also suggest you be sure that the camera is pointing toward useful information when the nodes start to be sure to be able to provide a valid depth.

dineshrboson, 2021-03-11:
[ INFO] [1615454209.255084811]: SDK version : 3.4.1
ROS MELODIC
[ INFO] [1615454270.697228300]: ZED connection -> SUCCESS
[ INFO] [1615454270.827610869]: rtabmap 0.20.7 started...
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[ INFO] [1615454271.321474653]: Reading parameters from the ROS server...
[ INFO] [1615454271.436991245]: Parameters read = 341
[ INFO] [1615454271.437016751]: Parameters successfully read.
[ INFO] [1615454271.727858872]: /zed/rtabmapviz: subscribe_depth = false
[ INFO] [1615454271.727894678]: /zed/rtabmapviz: subscribe_rgb = false
[ INFO] [1615454271.727904486]: /zed/rtabmapviz: subscribe_stereo = false
[ INFO] [1615454271.727917292]: /zed/rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1615454271.727925248]: /zed/rtabmapviz: subscribe_odom_info = false
[ INFO] [1615454271.727932894]: /zed/rtabmapviz: subscribe_user_data = false
[ INFO] [1615454271.727939805]: /zed/rtabmapviz: subscribe_scan = false
[ INFO] [1615454271.727946552]: /zed/rtabmapviz: subscribe_scan_cloud = false
[ INFO] [1615454271.727952758]: /zed/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1615454271.727959319]: /zed/rtabmapviz: queue_size    = 10
[ INFO] [1615454271.727966359]: /zed/rtabmapviz: approx_sync   = true
[ INFO] [1615454271.727989589]: Setup scan callback
[ INFO] [1615454271.729839403]: 
/zed/rtabmapviz subscribed to:
   /zed/zed_node/odom
[ INFO] [1615454271.729918365]: rtabmapviz started.
[ INFO] [1615454272.697752342]:  ...  ZED ready
[ INFO] [1615454272.697845618]: ZED SDK running on GPU #0
[ INFO] [1615454272.708291902]: Camera-IMU Transform: 
 55AD69AE5250
0.999947 0.010094 0.001764 -0.002000
-0.010097 0.999948 0.001617 -0.023000
-0.001748 -0.001635 0.999997 -0.002000
0.000000 0.000000 0.000000 1.000000

[ INFO] [1615454272.708428784]:  * CAMERA MODEL	 -> ZED 2
[ INFO] [1615454272.708546578]:  * Serial Number -> 26041082
[ INFO] [1615454272.708629526]:  * Camera FW Version -> 1523
[ INFO] [1615454272.708708223]:  * Sensors FW Version -> 776
[ INFO] [1615454272.878429793]: Advertised on topic /zed/zed_node/rgb/image_rect_color
[ INFO] [1615454272.878473219]: Advertised on topic /zed/zed_node/rgb/camera_info
[ INFO] [1615454272.890394524]: Advertised on topic /zed/zed_node/rgb_raw/image_raw_color
[ INFO] [1615454272.890489053]: Advertised on topic /zed/zed_node/rgb_raw/camera_info
[ INFO] [1615454272.902641102]: Advertised on topic /zed/zed_node/left/image_rect_color
[ INFO] [1615454272.902739660]: Advertised on topic /zed/zed_node/left/camera_info
[ INFO] [1615454272.919476830]: Advertised on topic /zed/zed_node/left_raw/image_raw_color
[ INFO] [1615454272.919590345]: Advertised on topic /zed/zed_node/left_raw/camera_info
[ INFO] [1615454272.933620184]: Advertised on topic /zed/zed_node/right/image_rect_color
[ INFO] [1615454272.933709390]: Advertised on topic /zed/zed_node/right/camera_info
[ INFO] [1615454272.946016122]: Advertised on topic /zed/zed_node/right_raw/image_raw_color
[ INFO] [1615454272.946053894]: Advertised on topic /zed/zed_node/right_raw/camera_info
[ INFO] [1615454272.958416714]: Advertised on topic /zed/zed_node/rgb/image_rect_gray
[ INFO] [1615454272.958471225]: Advertised on topic /zed/zed_node/rgb/camera_info
[ INFO] [1615454272.970341903]: Advertised on topic /zed/zed_node/rgb_raw/image_raw_gray
[ INFO] [1615454272.970378196]: Advertised on topic /zed/zed_node/rgb_raw/camera_info
[ INFO] [1615454272.982156011]: Advertised on topic /zed/zed_node/left/image_rect_gray
[ INFO] [1615454272.982196151]: Advertised on topic /zed/zed_node/left/camera_info
[ INFO] [1615454272.993814892]: Advertised on topic /zed/zed_node/left_raw/image_raw_gray
[ INFO] [1615454272.993855915]: Advertised on topic /zed/zed_node/left_raw/camera_info
[ INFO] [1615454273.006518759]: Advertised on topic /zed/zed_node/right/image_rect_gray
[ INFO] [1615454273.006608352]: Advertised on topic /zed/zed_node/right/camera_info
[ INFO] [1615454273.018351004]: Advertised on topic /zed/zed_node/right_raw/image_raw_gray
[ INFO] [1615454273.018397642]: Advertised on topic /zed/zed_node/right_raw/camera_info
[ INFO] [1615454273.030668358]: Advertised on topic /zed/zed_node/depth/depth_registered
[ INFO] [1615454273.030715687]: Advertised on topic /zed/zed_node/depth/camera_info
[ INFO] [1615454273.042685248]: Advertised on topic /zed/zed_node/stereo/image_rect_color
[ INFO] [1615454273.054762025]: Advertised on topic /zed/zed_node/stereo_raw/image_raw_color
[ INFO] [1615454273.055152347]: Advertised on topic /zed/zed_node/confidence/confidence_map
[ INFO] [1615454273.055495012]: Advertised on topic /zed/zed_node/disparity/disparity_image
[ INFO] [1615454273.056054291]: Advertised on topic /zed/zed_node/point_cloud/cloud_registered
[... 30 lines cut ...]
[ INFO] [1615454274.608650890]: Published static transform 'zed_imu_link' -> 'zed_left_camera_frame'
[ WARN] [1615454274.847833482]: Could not get transform from odom to base_link after 0.200000 seconds (for stamp=1615454274.593660)! Error="Lookup would require extrapolation into the past.  Requested time 1615454274.593660285 but the earliest data is at time 1615454274.605801209, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.201392 timeout was 0.2.".
[ WARN] (2021-03-11 14:47:55.089) MapsManager.cpp:984::publishMaps() Graph has changed! The whole cloud is regenerated.
[... cut ...]
[zed/zed_state_publisher-2] killing on exit
[ INFO] [1615454275.539832513]: rtabmapviz: ctrl-c catched! Exiting Qt app...
Saving parameters to /home/dineshr/.ros/rtabmap.cfg
rtabmap: Saving database/long-term memory... (located at /home/dineshr/.ros/rtabmap.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /home/dineshr/.ros/rtabmap.db, 1 MB)
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

dineshrboson, 2021-03-11:
latest zed ros wrapper and examples

dineshrboson, 2021-03-11:
its crashing continously

Myzhar, 2021-03-11:
OK, let's investigate this problem in deep. Can you start the ZED node normally without crashes?
Please try
`$ roslaunch zed_display_rviz display_zed2.launch`
an let me know if it works

dineshrboson, 2021-03-11:
working thanks

closed 2021-03-11 by dineshrboson