stereolabs/zed-ros-examples #28: [Question]

dineshrboson, 2021-03-18:
where is 2d occupancy grid map saved after running rtabmap launch file?
Is a way to create gmapping after running zed nodelet to change depth image to laserscan?
how to run zed map topics for only 2d occupancy grid mapping

Myzhar, 2021-03-18:
Hi @dineshrboson 

`where is 2d occupancy grid map saved after running rtabmap launch file?`
You must use the RTABmap gui to save the map. This can help:
https://github.com/introlab/rtabmap_ros/issues/301

`Is a way to create gmapping after running zed nodelet to change depth image to laserscan?`
Yes, we provide an example launch file for that:
https://github.com/stereolabs/zed-ros-examples/blob/master/examples/zed_nodelet_example/launch/zed_laserscan_nodelet.launch

`how to run zed map topics for only 2d occupancy grid mapping`
The ZED ROS Wrapper does not publish 2D occupancy grid topics, you must use external nodes like RTABmap and mapping to create them.

dineshrboson, 2021-03-18:
[image]
somehow I didnt get you correctly
my query is that where is the grid map saved in local machine
Saving parameters to /home/dineshr/.ros/rtabmap.cfg
rtabmap: Saving database/long-term memory... (located at /home/dineshr/.ros/rtabmap.db)
rtabmap: 2D occupancy grid map saved.
rtabmap: Saving database/long-term memory...done! (located at /home/dineshr/.ros/rtabmap.db, 100 MB)

when I ran that zed nodelet I get the laser scan from depth image but i am looking for gmapping with that laser scan topic some how I had done that with launch file included in that nodelet but when the obstacles are not showing in black colour in the map
but when tried with rtabmapping launch file and selecting only 2d mapping I am getting obstacles in black in colour in map

can I run only 2dmapping because of storage issue with rtabmapping as it takes lot of data.

Myzhar, 2021-03-18:
You must enable the 2D grid map in the preferences:
[image]

gmapping must be correctly configured to obtain the correct map:
http://wiki.ros.org/gmapping

dineshrboson, 2021-03-18:
[image]
2d map will be saved in yaml and pgm form right?

Myzhar, 2021-03-18:
If you need yaml and pgm you must start a map_server and then save the map when the mapping process is completed using the map_saver:
http://wiki.ros.org/map_server#Examples

dineshrboson, 2021-03-18:
ok kindly tell me a solution to run only gmapping with zed 
I would like to place a camera on my robot and teleop it and create a 2d occupancy grid map and then save it and make it navigated
for that I need 2d occupancy grid map if i run rtabmap navigation launch command and run for an hour the storage will be nearly 10gb so I dont want to save rtabmap but only 2doccupancy grid map

dineshrboson, 2021-03-18:
[image]
same map when ran with rtabmap command and another launch command

dineshrboson, 2021-03-18:
<node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">
    <param name="map_udpate_interval" value="2.0"/>
    <param name="delta" value="0.02"/>
    <param name="base_frame" value="base_link"/>
    <remap from="scan" to="/zed/scan"/>
    <remap from="map" to="/zed/map"/>
    
    <param name="minimumScore" value="150"/>
    <param name="linearUpdate" value="0.5"/>
    <param name="angularUpdate" value="0.436"/>
    <param name="temporalUpdate" value="-1.0"/>
    <param name="resampleThreshold" value="0.5"/>
    <param name="particles" value="20"/>
    <param name="xmin" value="-20"/>
    <param name="xmax" value="20"/>
    <param name="ymin" value="-20"/>
    <param name="ymax" value="20"/>
 </node> 

this code I wrote after zed nodelets launch code

dineshrboson, 2021-03-18:
<node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">
    <param name="map_udpate_interval" value="2.0"/>
    <param name="delta" value="0.02"/>
    <param name="base_frame" value="base_link"/>
    <remap from="scan" to="/zed/scan"/>
    <remap from="map" to="/zed/map"/>
    
    <param name="minimumScore" value="150"/>
    <param name="linearUpdate" value="0.5"/>
    <param name="angularUpdate" value="0.436"/>
    <param name="temporalUpdate" value="-1.0"/>
    <param name="resampleThreshold" value="0.5"/>
    <param name="particles" value="20"/>
    <param name="xmin" value="-20"/>
    <param name="xmax" value="20"/>
    <param name="ymin" value="-20"/>
    <param name="ymax" value="20"/>
 </node>

dineshrboson, 2021-03-18:
<node pkg="gmapping" type="slam_gmapping" name="slam_gmapping" output="screen">
    <param name="map_udpate_interval" value="2.0"/>
    <param name="delta" value="0.02"/>
    <param name="base_frame" value="base_link"/>
    <remap from="scan" to="/zed/scan"/>
    <remap from="map" to="/zed/map"/>

Myzhar, 2021-03-18:
It's normal... the rtabmap 2d map is obtained projecting the 3D points to the ground plane, it's not the same as using a 2D laser scan.

When you start the ZED nodelet that provides the virtual laser scan topic you can configure gmapping in the same way that you do it using a real laser scanner.
Explaining in details how to do that is out of the scope of this support because it requires too many details to be explained.
There are many guides on the web about the process, starting from the ROS wiki mapping page:
http://wiki.ros.org/gmapping
https://www.theconstructsim.com/exploring-ros-with-a-2-wheeled-robot-13-gmapping/
https://www.clearpathrobotics.com/assets/guides/kinetic/husky/HuskyGmapping.html

dineshrboson, 2021-03-18:
it worked thanks

dineshrboson, 2021-03-18:
Hi I am having a question like  can a robot navigate with rtabmap on realtime and drop off previously loaded map when the script is interupted and again same thing  without localizing 
robot ---> start navigation with live rtabmap formation --> detects obstacles -->stops or takes new route

dineshrboson, 2021-03-23:
is it possible to navigate a robot without mapping.......like detecting target person and take a path to reach him with motion planning algorithms in the middle of it if any objects are detected it should avoid them and take a new path like local planner

Myzhar, 2021-03-23:
@dineshrboson yes you can, you must write a ROS node that subscribes to depth data, processes them, and transforms the information in actions.
There are many algorithms to be used to perform such tasks, this is not the right place to list all of them, but you can find a lot of information googling for "robot point cloud obstacle avoidance".

yvzksgl, 2021-05-28:
Hi, whenever i install python3-rosdep2 because of "rosdep install --from-paths src --ignore-src -r -y"  instruction my rviz breaks. What should i do? Thanks.

closed 2022-04-27 by github-actions[bot]