stereolabs/zed-ros-examples #32: [Question]

yvzksgl, 2021-05-28:
Hi, whenever i install python3-rosdep2 because of "rosdep install --from-paths src --ignore-src -r -y"  instruction my rviz breaks. What should i do? Thanks.

yvzksgl, 2021-05-28:
my rosrun rviz rviz output:

[ INFO] [1622233579.621131732]: rviz version 1.14.7
[ INFO] [1622233579.621186345]: compiled against Qt version 5.12.8
[ INFO] [1622233579.621202846]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1622233579.633311777]: Forcing OpenGl version 0.
[ INFO] [1622233580.208522907, 4380.369000000]: Stereo is NOT SUPPORTED
[ INFO] [1622233580.208622713, 4380.369000000]: OpenGL device: GeForce GTX 1060 with Max-Q Design/PCIe/SSE2
[ INFO] [1622233580.208652676, 4380.370000000]: OpenGl version: 4,6 (GLSL 4,6).
/opt/ros/noetic/lib/rviz/rviz: symbol lookup error: /opt/ros/noetic/lib/librviz.so: undefined symbol: _ZN7tf2_ros17TransformListenerC1ERN3tf210BufferCoreERKN3ros10NodeHandleEbNS4_14TransportHintsE

Sekomer, 2021-09-13:
ahaşurasıdaboşmuş hıeıheıehe

Myzhar, 2021-09-13:
Hi, I'm sorry, but this seems a ROS issue and not a ZED wrapper issue.
I suggest you ask on ROS channels:
https://answers.ros.org/

closed 2021-09-13 by Myzhar