stereolabs/zed-ros-examples #35: [Question]  Simple base case USB camera

kevinlinxc, 2021-06-28:
Hi, I have the Zed camera and I would like to use it with an existing Ros melodic Ubuntu 18.04 ROS workspace. 

I want to be able to launch a launch file, and then be able to see the camera output on rqt_image_view. How can I achieve this?

Some more details:

- I have installed the ZED sdk and the CUDA dependency, although I do not have a GPU

- Running ZED Display shows the camera output (so it works) but that might not work with our networking infrasturcture, so rqt image view is kind of a litmus test for something that works for me.
- 
- Running zed_display_rviz creates nodes, but the gui shows no images and the nodes are not readable by rqt_image_view, presumablny bceause they are not the right datatype

Myzhar, 2021-06-28:
Hi @kevinlinxc
have you installed the `zed-ros-wrapper` for this Github repository: https://github.com/stereolabs/zed-ros-wrapper?

kevinlinxc, 2021-06-28:
Yes, my directory looks like this:
ls catkin_ws/src/cameras/
zed-ros-wrapper zed-ros-examples

Myzhar, 2021-06-29:
Hi @kevinlinxc , I'm really sorry, but I have missed this important fact in your first post: ` although I do not have a GPU`.
The ZED SDK cannot run without an Nvidia GPU running the CUDA processing and so the ZED ROS Wrapper cannot acquire any kind of information.
The `ZED_Explorer` tool can run without a GPU because it does not require depth processing and the stereo module is disabled for it.

You can use the ZED with ROS without a GPU by writing your own wrapper for the ZED Open Capture driver that does not require CUDA to retrieve image streams and sensors data: https://github.com/stereolabs/zed-open-capture

kevinlinxc, 2021-06-29:
I see, thank you. I'll give it a shot.

closed 2021-06-29 by kevinlinxc

kevinlinxc, 2021-07-06:
Alright, I copied and modified a usb camera launch file but it looks weird, and I think it's because there's two cameras. I'm wondering if you know how to fix this problem:

Launch file:
```
<launch>
  <arg name="videonum" />
  <node name="usb_cam$(arg videonum)" pkg="usb_cam" type="usb_cam_node" output="screen" >
    <param name="video_device" value="/dev/video$(arg videonum)" />
    <param name="image_width" value="640" />
    <param name="image_height" value="480" />
    <param name="pixel_format" value="yuyv" />
    <param name="camera_frame_id" value="usb_cam" />
    <param name="io_method" value="mmap"/>
  </node>
  <node name="image_view$(arg videonum)" pkg="image_view" type="image_view" respawn="false" output="screen">
    <remap from="image" to="/usb_cam$(arg videonum)/image_raw"/>
    <param name="autosize" value="true" />
  </node>
</launch>
```
Launch using
`roslaunch cameras camera_viewer.launch videonum:=2`
where cameras is the node name, camera_viewer.launch is the launch file, and videonum is found via ls /dev/video*

Example picture:
[image]

Myzhar, 2021-07-06:
@kevinlinxc Please do not reply to a closed issue. It's better if you create a new issue and you cite the closed in it with a link.

To use the ZED as a "simple webcam" I suggest you explore the ZED [Open Capture driver](https://github.com/stereolabs/zed-open-capture)... for example, you are using the wrong image resolution.