stereolabs/zed-ros-examples #36: [BUG]

joseandres15, 2021-07-22:
**Describe the bug**
When I run `roslaunch zed_rtabmap_example zed_rtabmap.launch`, the windows "zed-rtabmap.rviz*" and "RTAB-Map*" open. My problem comes with the "3D Map" display in "RTAB-Map*" window, which is frozen and I can't make it work. Also the "Odometry" display, which doesn't show anything, it keeps in gray.

Could you please tell me what is happening and how could I fix it?

**Device information:**
 - OS: Ubuntu
 - OS version: 18.04
 - CPU: Jetson AGX Xavier
 - GPU Nvidia:  [e.g. GTX1080, RTX1070]
 - ZED SDK Version: v.3.5
 - ROS Wrapper version: ROS Melodic
 - ROS rtabmap version: 0.20.9
 - ROS rtabmap_ros version: 0.20.9

**Screenshots**
[image]

Myzhar, 2021-07-22:
Hi @joseandres15 
have you tried to run it on a desktop PC? I just tried and everything works as expected. 
I suppose instead that the Nvidia Xavier AGX, even if it is very powerful, it is not powerful enough to handle simultaneously the RTABMAP processing, the Rviz displaying, and the RTABmap GUI displaying causing a freeze while visualizing the point cloud that is the information that requires the most of the processing.

joseandres15, 2021-07-22:
Hi @Myzhar,

No, I haven't tried it because I don't have a desktop PC right now.

My goal is to put the camera and the Jetson in a robot, in order it could scan and map the environment, am I able to do it with the jetson? Is there any other way to achieve this? Some way to do the rtabmapping in the Jetson Xavier?

Thank you.

Myzhar, 2021-07-22:
The Xavier AGX is powerful enough to handle the mapping process without any kind of issue.
You must instead run the GUIs to visualize the results on a remote computer to not waste resources for displaying.

joseandres15, 2021-07-22:
Okey, I will try it.

Thank you.

joseandres15, 2021-07-22:
Hi! I have a question, maybe a stupid question, but I need to know it.

Is there any possibility to develop this task in a Mac OS computer? I thought ZED SDK is only for Windows or Ubuntu, but I have seen this guy obtaining a point cloud with ZED in a Mac OS computer (as I can see in the images) and it would be really useful for me.

closed 2022-04-27 by github-actions[bot]