stereolabs/zed-ros-examples #38: relocation

JTDeve, 2021-07-25:
hi，i run the example and have stored the db file ,then i set localization = true in sl_rtabmap.launch.xml.But when i run ,the location seem worked,but the camera is in map center when start,and its toward is error.

Myzhar, 2021-07-26:
Hi @JTDeve 
can you explain your issue in detail?
What you are expecting, what you get instead, and your settings...

JTDeve, 2021-07-27:
@Myzhar thanks ,i have solved this problem,because i changed the zed_wrapper mapping default is true before.

JTDeve, 2021-07-31:
Hi @Myzhar 
I roslaunch zed_rtabmap_example zed_rtabmap.launch and get the rtabmap.db,
when I want to use location mode,I change the "sl_rtabmap.launch.xml" to set 'location=true',
then I roslaunch zed_rtabmap_example zed_rtabmap.launch ,I found the model is always in overlaps with the map_link,I want to get the location of the camera on the map, what should I do？

Myzhar, 2021-07-31:
Hi @JTDeve 
this is a question more related to RTABmap than to the ZED ROS Wrapper. 
I suggest you ask on the RTABmap forum:
http://official-rtab-map-forum.67519.x6.nabble.com/Rtabmap-mapping-done-but-can-t-do-relocalization-td1257.html

closed 2022-04-27 by github-actions[bot]