stereolabs/zed-ros-examples #43: Could not get transform from odom to base_link after 0.500000 seconds

Z-Jeff, 2022-06-15:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

When running `roslaunch zed_rtabmap_example zed_rtabmap.launch`, there occurs an error.
How can I solve this error?

> [ WARN] [1655283883.299755529]: Could not get transform from odom to base_link after 0.500000 seconds (for stamp=1655283881.908463)! Error="Lookup would require extrapolation into the future.  Requested time 1655283881.908463391 but the latest data is at time 1655283881.166989692, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.503115 timeout was 0.5.".

### Steps to Reproduce

`roslaunch zed_rtabmap_example zed_rtabmap.launch`

### Expected Result

No errors.

### Actual Result

The program  works well at begin.
But after a durantion,  the program doesn't work and an error occurs :
> [ WARN] [1655283883.299755529]: Could not get transform from odom to base_link after 0.500000 seconds (for stamp=1655283881.908463)! Error="Lookup would require extrapolation into the future.  Requested time 1655283881.908463391 but the latest data is at time 1655283881.166989692, when looking up transform from frame [base_link] to frame [odom]. canTransform returned after 0.503115 timeout was 0.5.".

### ZED Camera model

ZED2

### Environment

```shell
Hardware: NVIDIA TX2
ROS: ros-melodic
```

Myzhar, 2022-06-15:
Hi @Z-Jeff 
are you also showing the results of the processing on the TX2 or are you sending them to a remote Rviz?

Z-Jeff, 2022-06-15:
> 
I just run  `roslaunch zed_rtabmap_example zed_rtabmap.launch` on TX2, and nothing more.
It should only use the local Rviz in  zed_rtabmap.launch.

Myzhar, 2022-06-15:
Rviz is a high demanding application, summed to RTABmap and ZED SDK processing they push the TX2 over its limits, and the TF messages are so slowed down.
I suggest you do not run Rviz on the TX2, running it instead on a remote machine on the same local network.

closed 2022-08-22 by github-actions[bot]