stereolabs/zed-ros-examples #46: rviz maps basically nothing

khuechuong, 2023-01-20:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Basically, I am using a zed mini and using their ROS package. I use ubuntu 20.04, ros noetic, ros1, with a gtx 1080 gpu. I follow the installation instruction: https://www.stereolabs.com/docs/ros/
And I used rtabmap ros. In the beginning, both rtabmapviz and rviz wasn't that good as seen here.

RVIZ with  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

RVIZ without  _/zed/zed_node/point_cloud/cloud_registered_ topic:
[image]

So I **changed** the param _subscribe_scan_cloud_ to true and add <remap from="scan_cloud"        to="/zed/zed_node/point_cloud/cloud_registered"/> to rtabmap_ros and rtabmapviz. That really helped rtabmapviz, but rviz still just doesn't have the good quality of mapping like I see in other youtube. The command I ran was _roslaunch zed_rtabmap_example zed_rtabmap.launch_

**zed_rtabmap.launch**

<?xml version="1.0"?>

<launch>
    <arg name="svo_file"             default="" /> 
    <arg name="stream"               default="" /> 

    <arg name="zed_node_name"        default="zed_node" />
    <arg name="camera_model"         default="zedm" />
    <arg name="publish_urdf"         default="true" />

    <arg name="camera_name"          default="zed" />

    <arg name="base_frame"           default="base_link" />

    <group ns="$(arg camera_name)">
        
        <include file="$(find zed_wrapper)/launch/include/zed_camera.launch.xml">
            <arg name="camera_name"         value="$(arg camera_name)" />
            <arg name="svo_file"            value="$(arg svo_file)" />
            <arg name="stream"              value="$(arg stream)" />
            <arg name="node_name"           value="$(arg zed_node_name)" />
            <arg name="camera_model"        value="$(arg camera_model)" />
            <arg name="base_frame"          value="$(arg base_frame)" />
            <arg name="publish_urdf"        value="$(arg publish_urdf)" />
        </include>    
  
        
        <arg name="custom_rtabmap_launch_file" default="$(find zed_rtabmap_example)/launch/include/sl_rtabmap.launch.xml"/>
        <include file="$(arg custom_rtabmap_launch_file)">
            <arg name="odom_topic"              default="$(arg zed_node_name)/odom" />
            <arg name="rgb_topic"               default="$(arg zed_node_name)/rgb/image_rect_color" />
            <arg name="depth_topic"             default="$(arg zed_node_name)/depth/depth_registered" />
            <arg name="camera_info_topic"       default="$(arg zed_node_name)/rgb/camera_info" />
            <arg name="depth_camera_info_topic" default="$(arg zed_node_name)/depth/camera_info" />
        </include>   

        
        <node name="rviz" pkg="rviz" type="rviz" args="-d $(find zed_rtabmap_example)/rviz/zed-rtabmap.rviz" output="screen" /> 
    </group>
</launch>

**sl_rtabmap.launch.xml**

<launch>
    
    <arg name="localization"            default="false"/>

    <arg     if="$(arg localization)" name="args" default=""/>
    <arg unless="$(arg localization)" name="args" default="--delete_db_on_start"/>

[... 131 lines cut ...]

1.
2.
3.
...

### Expected Result

RVIZ mapping the same as rtabmpviz

### Actual Result

RVIZ maps just a few points and doesn't look anything like rtabmapviz

### ZED Camera model

ZED Mini

### Environment

```shell
GPU: GTX 1080 ti
ROS noetic
ros1
```

Myzhar, 2023-01-20:
Hi @khuechuong 
this is something related to the RTABmap package, not this example.
I suggest you contact RTABmap support:
http://official-rtab-map-forum.206.s1.nabble.com/

closed 2023-01-20 by Myzhar

matlabbe, 2023-01-29:
FYI: Tracking issue here: https://github.com/introlab/rtabmap_ros/issues/874